ros-industrial-attic/godel #38: Surface blending service dies

VictorLamoine, 2015-01-09:
I tested godel on different hardware configurations and on one configuration the node always crashes at start:
```
[surface_blending_service-2] process has died [pid 26385, exit code -11, cmd /home/meulage/catkin_ws/devel/lib/godel_surface_detection/surface_blending_service __name:=surface_blending_service __log:=/home/meulage/.ros/log/1b7dbaac-97dc-11e4-9637-7446a09160eb/surface_blending_service-2.log].
log file: /home/meulage/.ros/log/1b7dbaac-97dc-11e4-9637-7446a09160eb/surface_blending_service-2*.log

```
[logs](https://mega.co.nz/#!VUMhlbzZ!Cx68dR4FuyN7wLgKIenKTSA2PUBmDYvOqkZlIVBBA3o)

Any idea why this happens?

shaun-edwards, 2015-01-14:
@VictorLamoine, I need some more info.  Which node crashes?  What log message is output to the terminal (it's easier to parse this message than the log files you attached).

VictorLamoine, 2015-01-14:
Here is a `stdout` log example, the node `surface_blending_service` is crashing; RViz is launching but the `RobotBlendingPanel` panel is grayed out
```
meulage@HP-Compaq-Elite-8300-MT:~$ roslaunch godel_surface_detection robot_blending.launch
... logging to /home/meulage/.ros/log/ed99168c-9bcc-11e4-8dc4-7446a09160eb/roslaunch-HP-Compaq-Elite-8300-MT-2344.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://HP-Compaq-Elite-8300-MT:58167/

SUMMARY
========

PARAMETERS
 * /controller_joint_names: ['joint_s', 'join...
 * /initial_joint_state: [-1.34, 1.008, 0....
 * /mongo_wrapper_ros_HP_Compaq_Elite_8300_MT_2344_2433150109357235812/database_path: /home/meulage/cat...
 * /mongo_wrapper_ros_HP_Compaq_Elite_8300_MT_2344_2433150109357235812/overwrite: False
 * /move_group/allow_trajectory_execution: True
 * /move_group/allowed_execution_duration_scaling: 1.2
 * /move_group/allowed_goal_duration_margin: 0.5
 * /move_group/capabilities: move_group/MoveGr...
 * /move_group/controller_list: [{'action_ns': 'j...
 * /move_group/jiggle_fraction: 0.05
 * /move_group/manipulator/planner_configs: ['SBLkConfigDefau...
 * /move_group/manipulator_asus/longest_valid_segment_fraction: 0.05
 * /move_group/manipulator_asus/planner_configs: ['SBLkConfigDefau...
 * /move_group/manipulator_asus/projection_evaluator: joints(joint_s,jo...
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_simple_con...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_resolution: 0.025
 * /move_group/planner_configs/BKPIECEkConfigDefault/type: geometric::BKPIECE
 * /move_group/planner_configs/ESTkConfigDefault/type: geometric::EST
 * /move_group/planner_configs/KPIECEkConfigDefault/type: geometric::KPIECE
 * /move_group/planner_configs/LBKPIECEkConfigDefault/type: geometric::LBKPIECE
 * /move_group/planner_configs/PRMkConfigDefault/type: geometric::PRM
 * /move_group/planner_configs/PRMstarkConfigDefault/type: geometric::PRMstar
 * /move_group/planner_configs/RRTConnectkConfigDefault/type: geometric::RRTCon...
 * /move_group/planner_configs/RRTkConfigDefault/type: geometric::RRT
 * /move_group/planner_configs/RRTstarkConfigDefault/type: geometric::RRTstar
 * /move_group/planner_configs/SBLkConfigDefault/type: geometric::SBL
 * /move_group/planner_configs/TRRTkConfigDefault/type: geometric::TRRT
 * /move_group/planning_plugin: ompl_interface/OM...
 * /move_group/planning_scene_monitor/publish_geometry_updates: True
 * /move_group/planning_scene_monitor/publish_planning_scene: True
 * /move_group/planning_scene_monitor/publish_state_updates: True
 * /move_group/planning_scene_monitor/publish_transforms_updates: True
 * /move_group/request_adapters: default_planner_r...
 * /move_group/start_state_max_bounds_error: 0.1
 * /point_cloud_generator_node/frame_id: world_frame
 * /point_cloud_generator_node/point_cloud_descriptions: [{'h': 0.0, 'rx':...
 * /robot_description: <?xml version="1....
 * /robot_description_kinematics/manipulator/kinematics_solver: kdl_kinematics_pl...
 * /robot_description_kinematics/manipulator/kinematics_solver_attempts: 40
 * /robot_description_kinematics/manipulator/kinematics_solver_search_resolution: 0.01
 * /robot_description_kinematics/manipulator/kinematics_solver_timeout: 0.5
 * /robot_description_kinematics/manipulator_asus/kinematics_solver: kdl_kinematics_pl...
 * /robot_description_kinematics/manipulator_asus/kinematics_solver_attempts: 40
 * /robot_description_kinematics/manipulator_asus/kinematics_solver_search_resolution: 0.01
 * /robot_description_kinematics/manipulator_asus/kinematics_solver_timeout: 0.5
 * /robot_description_planning/joint_limits/joint_b/has_acceleration_limits: False
 * /robot_description_planning/joint_limits/joint_b/has_velocity_limits: True
 * /robot_description_planning/joint_limits/joint_b/max_acceleration: 0
 * /robot_description_planning/joint_limits/joint_b/max_velocity: 1.0
 * /robot_description_planning/joint_limits/joint_e/has_acceleration_limits: False
 * /robot_description_planning/joint_limits/joint_e/has_velocity_limits: True
 * /robot_description_planning/joint_limits/joint_e/max_acceleration: 0
 * /robot_description_planning/joint_limits/joint_e/max_velocity: 1.0
[... 216 lines cut ...]
	'/surface_detection'
	'/select_surface'
	'/surface_blending_parameters
[ INFO] [1421226529.149906620]: rviz blending panel connecting to services
[ INFO] [1421226529.151299812]: waitForService: Service [/surface_detection] has not been advertised, waiting...
================================================================================REQUIRED process [rviz-6] has died!
process has died [pid 2431, exit code -11, cmd nice /opt/ros/indigo/lib/rviz/rviz -d /home/meulage/catkin_ws/src/godel/godel_surface_detection/rviz/robot_blending.rviz __name:=rviz __log:=/home/meulage/.ros/log/ed99168c-9bcc-11e4-8dc4-7446a09160eb/rviz-6.log].
log file: /home/meulage/.ros/log/ed99168c-9bcc-11e4-8dc4-7446a09160eb/rviz-6*.log
Initiating shutdown!
================================================================================
[mongo_wrapper_ros_HP_Compaq_Elite_8300_MT_2344_2433150109357235812-11] killing on exit
[move_group-10] killing on exit
[robot_state_publisher-9] killing on exit
[joint_trajectory_action-8] killing on exit
[industrial_robot_simulator-7] killing on exit
[rviz-6] killing on exit
[polygon_offset_node-5] killing on exit
[process_path_generator_node-4] killing on exit
[point_cloud_generator_node-3] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

gavanderhoorn, 2015-01-14:
Can you run `surface_blending_service` in GDB? See [How to Roslaunch Nodes in Valgrind or GDB](http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB) for how to do this.

That would give us a stack trace, and possibly somewhere to begin searching.

It may be necessary to rebuild `surface_blending_service` with the `Debug` or `RelWithDebInfo` CMake profiles.

gavanderhoorn, 2015-01-14:
Not related, but just noticed this:

> the rosdep view is empty: call 'sudo rosdep init' and 'rosdep update'

You might want to take care of that (only needed once).

shaun-edwards, 2015-01-14:
In addition, you mentioned the node only crashes on one of multiple hardware configurations.  Out of curiosity is the one that is crashing a 32bit machine (vs. 64 bit)?

AnthonyBaup, 2015-01-14:
I work with VictorLamoine.

It crashes on a 64bits machine.
After using gdb it sems the problem comes from pcl :

[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff3fafc50 in boost::math::lanczos::lanczos_initializer<boost::math::lanczos::lanczos17m64, long double>::init::init() () from /usr/lib/libpcl_sample_consensus.so.1.7

PCL is in version 1.7.1-3 perhaps it is the problem ?

gavanderhoorn, 2015-01-14:
@AnthonyBaup: could you provide us with a backtrace (`bt <enter>` at the gdb prompt after it has crashed)?

AnthonyBaup, 2015-01-14:
@gavanderhoorn I tried to reinstall pcl from source (1.8) 
then I recompiled my workspace (godel CMakeList.txt has been modified to use PCL 1.8 and not 1.7 as before) and it worked.
There are no more errors using blending service.
Of cource, I don't have the segmentation fault using gdb anymore.

VictorLamoine, 2015-01-14:
@AnthonyBaup Please test with PCL release 1.7.2 as well (compile it from sources [here](https://github.com/PointCloudLibrary/pcl/releases/tag/pcl-1.7.2))

PCL 1.7.1 is not the current release, current is 1.7.2.
The error is on PCL side so I guess this can be closed?

gavanderhoorn, 2015-01-14:
Lets first make sure it was actually an ABI (?) issue.

VictorLamoine, 2015-01-14:
```
(gdb) bt
#0  0x00007ffff3fd9a70 in boost::math::lanczos::lanczos_initializer<boost::math::lanczos::lanczos17m64, long double>::init::init() ()
   from /usr/lib/libpcl_sample_consensus.so.1.7
#1  0x00007ffff3fb16be in ?? () from /usr/lib/libpcl_sample_consensus.so.1.7
#2  0x00007ffff7dea13a in call_init (l=<optimized out>, argc=argc@entry=1, argv=argv@entry=0x7fffffffdae8, env=env@entry=0x7fffffffdaf8) at dl-init.c:78
#3  0x00007ffff7dea223 in call_init (env=<optimized out>, argv=<optimized out>, argc=<optimized out>, l=<optimized out>) at dl-init.c:36
#4  _dl_init (main_map=0x7ffff7ffe1c8, argc=1, argv=0x7fffffffdae8, env=0x7fffffffdaf8) at dl-init.c:126
#5  0x00007ffff7ddb30a in _dl_start_user () from /lib64/ld-linux-x86-64.so.2
#6  0x0000000000000001 in ?? ()
#7  0x00007fffffffdea7 in ?? ()
#8  0x0000000000000000 in ?? ()
```

Our tests shows that using `libpcl-all` produces the seg fault, using `libpcl-1.7-all` solves it. (PCL trunk works too)

AnthonyBaup, 2015-01-14:
https://github.com/PointCloudLibrary/pcl/issues/619

VictorLamoine, 2015-01-14:
PointCloudLibrary/pcl#619

closed 2015-02-13 by VictorLamoine