ros-industrial-attic/godel #50: surface_detection: non-existing build_dependency on 'pcl'

gavanderhoorn, 2015-03-03:
This results in:
```
ERROR[godel_surface_detection]: Cannot locate rosdep definition for [pcl]
	rosdep key : pcl
	OS name    : ubuntu
	OS version : precise
	Data: <no data>
```

upon invoking `rosdep check ..`.

There is no ROS pkg or system dependency called `pcl`. I think the proper dependency would be `pcl_ros` (as that automatically pulls in `libpcl-all-dev` as well, on both Hydro and Indigo).

jrgnicho, 2015-03-05:
This will be addressed in PR #49 since this correction is needed in order to have a valid install procedure.

gavanderhoorn, 2015-03-10:
#49 was merged in ed26f1c and included the fix mentioned by @jrgnicho, so this can be closed.

closed 2015-03-10 by gavanderhoorn