ros-industrial-attic/godel #53: Installations instructions not complete

VictorLamoine, 2015-03-21:
```bash
~/MEGAsync/catkin_ws/src$ wstool merge https://github.com/ros-industrial-consortium/godel/raw/hydro-devel/godel.rosinstall
     Performing actions: 

     Add new elements:
  descartes,  godel,  godel_openvoronoi

Config changed, maybe you need run wstool update to update SCM entries.
Overwriting /home/victor/MEGAsync/catkin_ws/src/.rosinstall

update complete.
victor@desktop:~/MEGAsync/catkin_ws/src$ wstool update
[descartes] Fetching https://github.com/ros-industrial-consortium/descartes.git (version hydro-devel) to /home/victor/MEGAsync/catkin_ws/src/descartes
Clonage dans '/home/victor/MEGAsync/catkin_ws/src/descartes'...
remote: Counting objects: 1346, done.
remote: Compressing objects: 100% (36/36), done.
remote: Total 1346 (delta 10), reused 0 (delta 0), pack-reused 1308
Réception d'objets: 100% (1346/1346), 304.49 KiB | 0 bytes/s, done.
Résolution des deltas: 100% (822/822), done.
Vérification de la connectivité... fait.
[descartes] Done.
[godel_openvoronoi] Fetching https://github.com/ros-industrial-consortium/godel_openvoronoi.git (version hydro-devel) to /home/victor/MEGAsync/catkin_ws/src/godel_openvoronoi
Clonage dans '/home/victor/MEGAsync/catkin_ws/src/godel_openvoronoi'...
remote: Counting objects: 5581, done.
remote: Total 5581 (delta 0), reused 0 (delta 0), pack-reused 5581
Réception d'objets: 100% (5581/5581), 4.61 MiB | 1.39 MiB/s, done.
Résolution des deltas: 100% (3369/3369), done.
Vérification de la connectivité... fait.
[godel_openvoronoi] Done.
[godel] Fetching https://github.com/ros-industrial-consortium/godel.git (version hydro-devel) to /home/victor/MEGAsync/catkin_ws/src/godel
Clonage dans '/home/victor/MEGAsync/catkin_ws/src/godel'...
remote: Counting objects: 2899, done.
remote: Total 2899 (delta 0), reused 0 (delta 0), pack-reused 2899
Réception d'objets: 100% (2899/2899), 14.85 MiB | 1.51 MiB/s, done.
Résolution des deltas: 100% (1662/1662), done.
Vérification de la connectivité... fait.
[godel] Done.
victor@desktop:~/MEGAsync/catkin_ws/src$ rosdep install --from-paths . --ignore-src
WARNING: Package "ompl" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
descartes_trajectory: Cannot locate rosdep definition for [console_bridge]
fermi_irb2400_moveit_config: Cannot locate rosdep definition for [abb_common]
descartes_planner: Cannot locate rosdep definition for [console_bridge]
descartes_core: Cannot locate rosdep definition for [console_bridge]
```

```bash
~/MEGAsync/catkin_ws/src$ wstool 
workspace: /home/victor/MEGAsync/catkin_ws/src

 Localname         S SCM  Version-Spec UID  (Spec)  URI  (Spec) [http(s)://...]
 ---------         - ---- ------------ -----------  ---------------------------
 godel               git  hydro-devel  43ff64fcb528 github.com/ros-industrial-consortium/godel.git
 godel_openvoronoi   git  hydro-devel  32d71e35fe10 github.com/ros-industrial-consortium/godel_openvoronoi.git
 descartes           git  hydro-devel  d366700741d4 github.com/ros-industrial-consortium/descartes.git

Also detected these repositories in the workspace, add using 'wstool set':

 Localname               SCM   URI [http(s)://...]
 ---------               ----  -------------------
 fanuc                   --git git@github.com:VictorLamoine/fanuc.git
 fanuc_experimental      --git git@github.com:VictorLamoine/fanuc_experimental.git
 fanuc_m10ia_ensenso_eef --git git@github.com:VictorLamoine/fanuc_m10ia_ensenso_eef.git
 fanuc_m10ia_kinect_eef  --git 
 fermi                   --git git@github.com:VictorLamoine/fermi.git
 industrial_core         --git github.com/ros-industrial/industrial_core.git
 motoman                 --git github.com/ros-industrial/motoman.git

```

VictorLamoine, 2015-03-21:
Also, QD is still needed; (`libqd-dev`)

```bash
odel_openvoronoi/src)
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   graph
CMake Error at godel_openvoronoi/src/Findqd.cmake:39 (MESSAGE):
  Could not find QD
Call Stack (most recent call first):
  godel_openvoronoi/src/CMakeLists.txt:9 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/victor/MEGAsync/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/victor/MEGAsync/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```

gavanderhoorn, 2015-03-21:
Because of the error while trying to resolve `console_bridge`, `rosdep` hasn't installed all dependencies. That is why `libqd-dev` hasn't been installed (and potentially many others neither).

Can you try `rosdep --skip-keys=console_bridge install ..`? That should let it continue with the rest of the dependencies.

gavanderhoorn, 2015-03-21:
> fermi_irb2400_moveit_config: Cannot locate rosdep definition for [abb_common]

Also, this pkg is not part of the `godel` repository. It's rather strange that `rosdep` isn't able to resolve that dependency, because it has been released.

gavanderhoorn, 2015-03-21:
In general, to get `rosdep` to continue if it encounters errors, you can use the `-r` option.

gavanderhoorn, 2015-03-21:
Ah. On an Indigo system, the ABB pkgs haven't been released, so you'll get that error. `console_bridge` seems to have been released though?

VictorLamoine, 2015-03-23:
We got it working by skipping the missing dependencies :+1: 
In short we just changed 
`rosdep install --from-paths . --ignore-src`
to
`rosdep install --from-paths . --ignore-src -r`

Should I close this issue? It's clearly `indigo` related

shaun-edwards, 2015-03-23:
We will keep it open to address in Indigo.  This way we won't forget.

gavanderhoorn, 2015-03-23:
> Should I close this issue? It's clearly `indigo` related

I'm not so sure. If we ignore the `abb_common` issue (which is not part of Godel), only `console_bridge` seems to give you problems. Afaict, it has been released in Indigo, so I don't get why you can't resolve it using `rosdep`.

What does `rosdep resolve console_bridge` give you (after a `rosdep update`)?

VictorLamoine, 2015-03-23:
Neither of them can be resolved:
```
$ rosdep resolve console_bridge
ERROR: no rosdep rule for 'console_bridge'
```

```
$ rosdep resolve abb_common
ERROR: no rosdep rule for 'abb_common'
```

gavanderhoorn, 2015-03-23:
As I wrote earlier, the `abb_common` one is to be expected, as it has not been released yet on Indigo.

`console_bridge` does actually not exist (anymore) on Indigo, so that should be fixed. I've created a separate issue for that.

@VictorLamoine: can you confirm that the `console_bridge` key was the only one giving you trouble on your Indigo system?

gavanderhoorn, 2015-03-23:
Hm just checked this myself. The following pkgs also couldn't be resolved (but expected, as they haven't been released yet): `simple_message` (and all dependencies), `motoman_sia20d_support` (and all dependencies).

For now a clone of `motoman` and `industrial_core` should let users continue.

@jrgnicho, @shaun-edwards: perhaps those pkgs should be temporariliy added to `godel.rosinstall`? Or do we want to wait for the Indigo release of `industrial_core` and `motoman`?

VictorLamoine, 2015-03-23:
> can you confirm that the console_bridge key was the only one giving you trouble on your Indigo system?

Yes

> For now a clone of motoman and industrial_core should let users continue

I've not seen that because I already have clones of `industrial_core` (for `fanuc`) and `motoman` (for `fermi`)

jrgnicho, 2015-03-23:
Where the **console_bridge** dependency errors coming from any godel packages or from others repos listed in the .rosinstall file (descartes and godel_openvoronoi) ?

gavanderhoorn, 2015-03-23:
Just retested, no pkgs in `godel` (or any of its direct dependencies) depend on `console_bridge` anymore (after ros-industrial-consortium/descartes#72). So I guess @VictorLamoine was using an out-of-date clone of `descartes`?

VictorLamoine, 2015-03-23:
> So I guess @VictorLamoine was using an out-of-date clone of descartes?

I don't think so; I've done a `wstool update`

I've just tested in a clean workspace and it works fine (I manualy cloned `motoman` and `industrial_core`)

jrgnicho, 2015-03-24:
I suppose that `wstool update` might have not worked correctly due to existing artifacts in the workspace.  I've haven't seen it fail this way but then again I'm not that well verse in **wstool** myself.

gavanderhoorn, 2015-03-24:
@jrgnicho: could be that `wstool` won't touch existing clones of repositories. @VictorLamoine: did you already have a manual clone of `descartes` in the problematic workspace?

VictorLamoine, 2015-03-24:
> @VictorLamoine: did you already have a manual clone of descartes in the problematic workspace?

That is possible unfortunately I deleted everything `godel` related to test from zero so I don't have this info anymore.

As said just before the only thing missing are the `motoman` and `industrial_core` packages; which I have in my workspace to compensate.

Everything works fine with the instructions now except the fermi / abb_common dependency; but it has nothing to do with `godel`. (and the `-r` option solves the problem).

:+1:

jrgnicho, 2015-03-25:
Can this issue be closed?

VictorLamoine, 2015-03-25:
> We will keep it open to address in Indigo. This way we won't forget.

Until the motoman` and `industrial_core` "problems" are solved, we'll keep this issue open as a reminder.

jrgnicho, 2015-03-25:
Perhaps we should close this one a create a new one that describes the indigo dependencies issue more specifically?

gavanderhoorn, 2015-03-30:
I agree with @jrgnicho: it would be better to create a new issue.

For the current one, we might want to consider updating the instructions in the `README.md` with some of the comments posted (ie: what to do for missing `rosdep` keys (non-released pkgs fi)). Question is: how much should we fool-proof the instructions?

closed 2015-03-30 by VictorLamoine