ros-industrial-attic/godel #54: console_bridge not a valid dependency in Indigo

gavanderhoorn, 2015-03-23:
As reported in #53, `rosdep` cannot resolve `console_bridge` to any ros pkg or system library.

If this dependency is still needed for Godel, we should probably change the dependency to `libconsole-bridge-dev` (as suggested by [wiki/console_bridge](http://wiki.ros.org/console_bridge)).

jrgnicho, 2015-03-23:
The error log in #53 only showed unresolved console_bridge dependencies from descartes.  This was already addressed [here](https://github.com/ros-industrial-consortium/descartes/pull/72) .  So there shouldn't be any **console_bridge** dependencies there as far as I know.  I don't think that godel depends directly on **console_bridge** either.

gavanderhoorn, 2015-03-23:
I know PR#72 on Descartes, but I encountered the same errors while testing out the build instructions on Indigo. That's why I reported the issue. I'll test this again later and report back.

gavanderhoorn, 2015-03-23:
My only excuse is that it's already 'late' here :). No `rosdep check ..` errors about `console_bridge` in a clean workspace at all.

Sorry for the noise.

closed 2015-03-23 by gavanderhoorn

jrgnicho, 2015-03-24:
Not a problem.  However it did expose dependency issues (industrial_core and motoman) which will have to be addressed when migrating to indigo.

gavanderhoorn, 2015-03-24:
Yes. Well `industrial_core` is being released into Indigo, so that should not pose any problems anymore.

The `motoman` pkgs won't either, as soon as they are released. For now we could think about adding `motoman` to `godel.rosinstall`. We can always remove it once `motoman` is released.

Only potential issue could be with users that do the setup once -- and then have a `motoman` clone in the godel ws that always shadows the released one.