ros-industrial-attic/godel #55: Surface selection issue

FrancoisLasson, 2015-03-24:
Beginner with Godel,
I'm trying to use the godel_surface_detection package.

I launch : $ roslaunch godel_surface_detection robot_blending.launch
and I try to « Scan and Find » some surfaces.

Even though "Selected surface clusters" are founded, I don't succeed in select these surfaces. 
Could you explain me the solution to select them?

log files : https://mega.co.nz/#!BYAVSZha!zYmWa4sig1CgIwDRZehJvpxYjebqeMSbtc7cr1JVeTs

Thank you, 
François.

Jmeyer1292, 2015-03-25:
I'll look into this issue today.

Jmeyer1292, 2015-03-25:
There's a missing return statement on here:
https://github.com/ros-industrial-consortium/godel/blob/hydro-devel/godel_surface_detection/src/detection/surface_detection.cpp#L833-L834

It should have "return true" before the closing brace of the function. Try it out and let me know. This will be fixed in pull requests I'm working on right now.

FrancoisLasson, 2015-03-26:
Your solution fixes the problem. From now on, all surfaces can be select. 
Thank you.

Jmeyer1292, 2015-12-21:
@jrgnicho This should be closed.

Jmeyer1292, 2015-12-21:
Addressed by #65. Thank you for brining this issue to light.

jrgnicho, 2015-12-21:
Address in #65

closed 2016-01-07 by Jmeyer1292