ros-industrial-attic/godel #62: Update installation instructions for jade

sketchc89, 2015-06-16:
Related #53 #58 

`rosdep install --from-paths . --ignore-src -r`

Gets past qd dependency like in #53 but move-it dependencies cause errors in godel path planning
```
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install ros-jade-moveit-ros-visualization] failed
  apt: command [sudo -H apt-get install ros-jade-moveit-ros-planning-interface] failed
  apt: Failed to detect successful installation of [ros-jade-moveit-ros-visualization]
  apt: Failed to detect successful installation of [ros-jade-moveit-ros-planning-interface]
```

Reverting back to indigo since it looks like jade-moveit is still in development. Issue posted to log dependency issues for jade more specifically.

gavanderhoorn, 2015-06-16:
According to [debbuild/jade?q=moveit](http://www.ros.org/debbuild/jade?q=moveit), both `moveit_ros_visualization` and `moveit_ros_planning_interface` seem to have been released for Jade.

Looking at the status 'blocks' though, it looks like only the Trusty targets have been built successfully. On what OS are you running the `rosdep install ..` command?

sketchc89, 2015-06-17:
Ubuntu 15.04 vivid. I've since reinstalled 14.04 for stability. I didn't read Indigo wasn't supported on 15.04 before installing it. It may be the packages weren't available for 15.04 yet. I couldn't find them in the ppa on the (Jade install page)[http://wiki.ros.org/jade/Installation/Ubuntu].

gavanderhoorn, 2015-06-17:
@sketchc89 wrote:
> Ubuntu 15.04 vivid. 

ah k. That would seem to explain the `rosdep` output then.

> I've since reinstalled 14.04 for stability. I didn't read Indigo wasn't supported on 15.04 before installing it. It may be the packages weren't available for 15.04 yet. I couldn't find them in the ppa on the [Jade install page](http://wiki.ros.org/jade/Installation/Ubuntu).

Yes, Indigo is supported on Trusty and Saucy (see also [REP-003](http://www.ros.org/reps/rep-0003.html#indigo-igloo-may-2014)) only. For Jade, those pkgs are indeed missing at the moment. It might perhaps be an idea to report this to the maintainers of those pkgs?

sketchc89, 2015-06-17:
Yeah, that'll work.

phvass, 2016-10-28:
Seems like we are planning to skip Jade and move to Kinetic next.

closed 2016-10-28 by phvass