ros-industrial-attic/godel #97: Indigo  rosdep error when execute rosdep install --from-paths . --ignore-src

chenchengshao, 2016-01-24:
Version: unbutu 14.04 trusty   Indigo ROS system.
Command:  rosdep install --from-paths . --ignore-src
Return:
WARNING: Package "ompl" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
godel_robot_resources: Cannot locate rosdep definition for [industrial_extrinsic_cal]

Jmeyer1292, 2016-01-25:
Thanks for bringing this to my attention, I'll add the appropriate dependency on industrial_extrinsic_cal. In the mean-time, if you clone the`indigo-devel` branch of [this repo](https://github.com/ros-industrial/industrial_calibration.git), you should be able to make progress.

Jmeyer1292, 2016-01-25:
@chenchengshao We have elected to forgo the dependency on industrial calibration at the moment to ease the installation process for those just getting into the project. See #98. When this is accepted, we should be good.

chenchengshao, 2016-01-25:
Thank you Jmeyer1292, I already tried to use rosdep install -r --from-paths . --ignore-src to ignore this error and also install the abb separately, it should have the same result as the new solution you posted.

closed 2016-01-25 by Jmeyer1292