ros-industrial-attic/godel #99: Missing line in CMakeLists.txt causes packages to not build with catkin tools.

jettan, 2016-02-12:
Offending files are:
```
godel_robots/abb/godel_irb2400/godel_irb2400_support/CMakeLists.txt
godel_robots/godel_robot_resources/CMakeLists.txt
```

The missing line:

```
find_package(catkin REQUIRED)
```

closed 2016-02-14 by Jmeyer1292

jettan, 2016-02-14:
Still missing the line in godel_robots/godel_robot_resources/CMakeLists.txt

gavanderhoorn, 2016-02-14:
Pretty sure I added it [here](https://github.com/ros-industrial-consortium/godel/blob/4d3025dd974ad48a05cd1575d6e8ad096be9b2d0/godel_robots/godel_robot_resources/CMakeLists.txt#L4).

jettan, 2016-02-14:
Ah,  my bad then. Didn't see it since 5321918 didn't mention it.

gavanderhoorn, 2016-02-14:
I opted to fix each file in a separate commit. So that's why.

gavanderhoorn, 2016-02-14:
Thanks for reporting.

commit 5321918f (2016-02-14): irb2400_support: add missing find_pkg(catkin). Fix #99.