ros-industrial-attic/godel #105: catkin make error

james2254, 2016-04-05:
When I run catkin_make in the terminal ,there is a error 。“ 
+++ processing catkin package: 'godel_openvoronoi'
-- ==> add_subdirectory(godel_openvoronoi/src)
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   graph
CMake Error at godel_openvoronoi/src/Findqd.cmake:39 (MESSAGE):
  Could not find QD
Call Stack (most recent call first):
  godel_openvoronoi/src/CMakeLists.txt:9 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/james/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/james/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
”
what's happen ?how can I correct it?

gavanderhoorn, 2016-04-05:
Did you follow the installation instructions in the [readme](https://github.com/ros-industrial-consortium/godel/blob/indigo-devel/README.md)?

The error you see is probably caused by a missing `libqd-dev` (needed by `godel_openvoronoi`), which the `rosdep install ..` step in the installation instructions installs for you.

james2254, 2016-04-06:
Thank you , it's correct follow the installation.

phvass, 2017-01-13:
Assuming this was just an installation issue... Closing.

closed 2017-01-13 by phvass