ros-industrial-attic/godel #110: Edge Processing

phvass, 2016-10-28:
- [ ] `Create edge detection procedure for captured clouds`
- [ ] `Develop edge detection planner`
- [ [ `Develop and implement quality metrics for edge processing`
- [ ] `GUI integration of edge-processor`

Jmeyer1292, 2017-01-27:
With #123 and subsequent PRs we have made a lot of progress toward this goal.

We still need:
 - Edge inspection routines
 - GUI integration (into the simple-gui?)
 - Some refinements to edge joint planning to control for things like the density of boundary points, noise in the data, and any process adjustments (such as a tool angle) we might need.

phvass, 2017-02-17:
Edge inspection is not in-scope for the M4 FTP. Simple-GUI integration is already captured in issue #158.

closed 2017-02-17 by phvass