ros-industrial-attic/godel #111: Macro-Scan Changes

phvass, 2016-10-28:
- [ ] `Add point-cloud into collision model`
- [ ] `Extract more complex surfaces (sloping surfaces with extrusions or holes)`
- [ ] `Recognize "holes" in the macro scan data`

phvass, 2016-11-08:
@gavanderhoorn - The first item in this list, adding point clouds to the collision model is in need of early attention to support our upcoming demo. Is that something you could help with?

gavanderhoorn, 2016-11-25:
Looks like this should be possible using the normal octomap updater plugin for MoveIt (like [this](https://github.com/ros-industrial-consortium/godel/compare/camera_demo...gavanderhoorn:add_octomap_moveit)) + updating the planning scene at the appropriate times in Descartes (before planning is started, so `MoveitStateAdapter::isInCollision(..)` uses an up-to-date scene).

@Jmeyer1292: does that make sense to you?

Three questions:

 1. should the octomap be continously updated, or only at certain times (ie: triggered by a service)?
 1. will the pointclouds that will be input to the octomap updater already be filtered, or will they contain the raw data coming from the rgbd sensor?

These questions are related, because if the pointcloud is already filtered, then we can probably get away with continuous updates (as only the workpiece geometry will be in there). For performance reasons, we could still go for the single publication approach, though. That would only update the octomap at certain times. Not updating continuously probably also removes the need for self-filtering (which can have quite a performance impact as well).

gavanderhoorn, 2016-11-29:
I've got a minimally working implementation. It wasn't entirely clear to me how much change would be acceptable, so the approach I took is not too clean, but perhaps Descartes would need to be changed for that a bit which seems out of scope (at least for now).

See https://github.com/ros-industrial-consortium/godel/compare/camera_demo...gavanderhoorn:add_octomap_moveit_descartes for the Godel diff. Depends on https://github.com/ros-industrial-consortium/descartes/compare/indigo-devel...gavanderhoorn:update_ps for the Descartes extension.

To test:

```
$ roslaunch godel_irb2400_support irb2400_blending.launch
```

That should start the default Godel blending demo. Add a *Planning Scene* display to RViz, then:

```
$ roslaunch godel_process_planning pub_cat.launch
```

This should result in a small cat being inserted, and the planning scene updated with the voxel representation.

After that use the demo as normal.

Planning for the largest plane should either fail completely, or just for the `surface_marker_server_1_scan` trajectory.

gavanderhoorn, 2016-11-29:
Note: I'm half-assuming here that point clouds fed to the octomap updater will have already been filtered / segmented. It should work if they are not, but that'll probably be slower.

gavanderhoorn, 2016-11-30:
Note also that I had to change `abb_irb2400_descartes` and make the `getAllIk(..)` override in it call `isValid(..)`, instead of doing the bounds check itself. As IK-solving and collision checking are done in the same place/phase, I don't see a way around that without changing how Descartes works.

gavanderhoorn, 2016-11-30:
See #119 and ros-industrial-consortium/descartes#187.

phvass, 2017-03-03:
This first item on the list is now tracked by #119 . The second item is already implict in the curved surface segmentation process. Closing this issue.

closed 2017-03-03 by phvass