ros-industrial-attic/godel #114: Line scanner calibration

phvass, 2016-10-28:
Calibrate the laser line scanner to the robot.

VictorLamoine, 2016-11-10:
If you mean extrinsic calibration -> adding a cost function into `industrial_calibration` I'm very interested in helping you (writing code, testing etc.).

We plan to buy a laser line scanner in the beginning of next year, it is not 100% certain yet.

phvass, 2017-01-13:
Tentatively assigning to Victor

VictorLamoine, 2017-01-18:
2 good news:
- I have the two profilometers
[image]
- We receive a Fanuc M-710iC/50 robot that I will able to use next week

I don't think I'll be able to start working on this before mid/end February

phvass, 2017-01-18:
Nice! Envious of the fact that you have two to choose from ;)

Paul Hvass
Program Manager
ROS-Industrial Consortium Americas
SwRI Robotics and Automation Engineering
210.522.5823 - W
210.896.0270 - M
Paul.Hvass@swri.org<mailto:Paul.Hvass@swri.org>
ROSindustrial.org

[materiel recu]<https://cloud.githubusercontent.com/assets/5566160/22061305/a46ec812-dd74-11e6-9d0b-4675f0ba9148.jpg>

phvass, 2017-03-02:
Any update on this @VictorLamoine ?

VictorLamoine, 2017-03-03:
I've just created an issue there:
https://github.com/ros-industrial/industrial_calibration/issues/95