ros-industrial-attic/godel #117: RViz crashes often on starting 'irb2400_blending.launch'

gavanderhoorn, 2016-11-28:
Vague, I admit, but often when starting the whole system in demo mode using `roslaunch godel_irb2400_support irb2400_blending.launch` I end up with the following:

```
[1480368908.987856538] [ INFO] [/rviz] [ros.godel_simple_gui]: Loaded simple blending panel
[1480368909.027215174] [ INFO] [/rviz] [ros.godel_simple_gui]: Initializng simple blending panel
================================================================================REQUIRED process [rviz-15] has died!
process has died [pid 12058, exit code -11, cmd nice /opt/ros/indigo/lib/rviz/rviz -d /home/user/ros/godel_ws/src/godel/godel_robots/abb/godel_irb2400/godel_irb2400_support/rviz/irb2400_blending.rviz __name:=rviz __log:=/home/user/.ros/log/8304f744-b5b2-11e6-adeb-d8d385994de6/rviz-15.log].
log file: /home/user/.ros/log/8304f744-b5b2-11e6-adeb-d8d385994de6/rviz-15*.log
Initiating shutdown!
================================================================================
```

Haven't had time to look into it, but `-11` (`SEGFAULT`) seems to point to a potential issue with one of the RViz panel plugins.

shaun-edwards, 2016-11-28:
Is this a Qt5 vs Qt4 issue?  We ran into the same problem when upgrading to kinetic.

gavanderhoorn, 2016-11-29:
No. This is all on Indigo. I don't have Qt5 installed.

gavanderhoorn, 2016-11-29:
Additionally: it does sometimes work. If this were a library incompatibility it should always crash.

Jmeyer1292, 2016-11-29:
It's a known issue. The blending panels frequently crash at start up. No one has looked into it yet.

Try again until it works.

gavanderhoorn, 2016-11-29:
Figured.

I've been running the `rviz` node with a gdb `launch-prefix` for a day now: as expected it hasn't happened yet / any more ..

phvass, 2017-03-02:
Crashing issue not occurring on Automate Demo computer. Changed milestones.