ros-industrial-attic/godel #126: Multicore Enhancement for Normal Estimation (OpenMP)

woodaaron, 2017-01-16:
Per the [PCL Normal Estimation Tutorial](http://pointclouds.org/documentation/tutorials/normal_estimation.php),
We can speed up the normal estimation process on multicore systems by using the OpenMP implementation of NormalEstimation. This might be useful in the godel surface segmentation library.

shaun-edwards, 2017-01-16:
Since the API is the same, seems like a no-brainer ;)  Good find!  

Glad to see you working on this @woodaaron.

gavanderhoorn, 2017-01-16:
Something to consider: without some rate limiting / setup, OpenMP can easily saturate all the cores on a machine, especially with parallel for-loops and similar constructs.

Jmeyer1292, 2017-01-17:
I don't believe saturation is an issue here. We don't run anything time sensitive in parallel with this operation.

gavanderhoorn, 2017-01-17:
You might not now. I just believe it makes sense to think ahead, and consider potential use of this code in other applications.

Jmeyer1292, 2017-01-17:
Ok, that's fair. Still, all of the recently added code has slowed things down significantly and if @woodaaron has measured and determined this to be an important optimization point I'm still on board.

@gavanderhoorn How would you deal with the saturation issues you mentioned? Control the size of the thread pool via some compile time param?

gavanderhoorn, 2017-01-17:
I haven't looked at the approach that PCL uses, but there are multiple ways. One is [OMP_NUM_THREADS](https://gcc.gnu.org/onlinedocs/libgomp/OMP_005fNUM_005fTHREADS.html#OMP_005fNUM_005fTHREADS), another would be [omp_set_num_threads(..)](https://gcc.gnu.org/onlinedocs/libgomp/omp_005fset_005fnum_005fthreads.html#omp_005fset_005fnum_005fthreads).

If not configured otherwise, OpenMP will use `OMP_NUM_THREADS == num_cpus`. For tight loops that can bring systems to their knees pretty quick.

woodaaron, 2017-01-17:
pcl::NormalEstimationOMP has a [setNumberOfThreads](http://docs.pointclouds.org/1.7.0/classpcl_1_1_normal_estimation_o_m_p.html#a88d42957ef34d3aeaf51dde07c53688d) function we could use.

Does OpenMP's [omp_set_dynamic](https://msdn.microsoft.com/en-us/library/3aw0bk98.aspx) take care of this issue for us? I'm a little confused how we would _a priori_ adjust for a user's unique use_case. Provide a parameter in our library constructor?

Jmeyer1292, 2017-01-18:
Addressed in #129

closed 2017-01-18 by Jmeyer1292