ros-industrial-attic/godel #147: Instructions for Keyence

woodaaron, 2017-02-03:
#142 incorporated keyence_experimental and relies on the install-libsocket++.sh script. @Jmeyer1292 , we should include instructions to run the script when setting up the Godel project. Failure to do so will lead to a build error

`fatal error: libsocket/inetclientstream.hpp: No such file or directory`

I didn't notice it when reviewing the pull request as the machine I was working on already had the library installed.

Jmeyer1292, 2017-02-03:
Good idea. Sorry I overlooked it as well. I'll make a PR that indicates the new requirement.

gavanderhoorn, 2017-02-03:
If I may suggest: build the library once on a clean install (or a Docker container) and use `checkinstall` to create a `.deb`. Then distribute that.

Installing things from sources is never nice on systems that are managed by a package manager.

Alternatively: perhaps @Levi-Armstrong can help you set up a PPA and to create a proper `.deb`. That would significantly ease deployment.

Jmeyer1292, 2017-02-03:
@gavanderhoorn That's how I already build all of my code, so I've got the `deb`. How would distribution work (short of creating a PPA)? Would we include it in this repo or host it somewhere separately?

gavanderhoorn, 2017-02-03:
Just building it on a random install (ie: 'polluted' by normal use) is probably something to avoid.

I'm pretty sure you can't import a checkinstall deb to a PPA: there are actually quite stringent constraints on form and content of PPAs and the debs in them.

Adding it to the repository as a binary is also an option.

Levi-Armstrong, 2017-02-05:
I can set-up a ppa for installing this if you would like. It may be a good point to talk about setting up one launchpad.net account for ROS-I items like this. I could create a new account for ROS-I and create a team if you think that is better?

Jmeyer1292, 2017-02-08:
Closed per #148. I'm not worried about the debian at this time.

closed 2017-02-08 by Jmeyer1292