ros-industrial-attic/godel #151: Edge Path Density

Jmeyer1292, 2017-02-09:
Our edge detection code returns a pose for every single point that is considered to be on a boundary. This means, especially for point clouds assembled from multiple point clouds, that these paths are very dense. This:
 - Puts a strain on motion planning (using Descartes at least)
 - Can cause the robot (IRB2400) to dwell

We could
 - Accept the density and try to focus on improving the motion planning smarts
 - Try to replace the dense point representation with something like a sequence of polynomials and then sample those at some controlled spacing.

sketchc89, 2017-02-13:
> Try to replace the dense point representation with something like a sequence of polynomials and then sample those at some controlled spacing.
There are a number of ways to simplify these dense paths summarized nicely in this [article](https://www.codeproject.com/Articles/114797/Polyline-Simplification). You would be losing some information about the geometry of the part, but this seems like it would be be a good trade-off to make since many machining tools can't take advantage of these very dense paths.

Jmeyer1292, 2017-02-13:
@sketchc89 That's an awesome link.

shaun-edwards, 2017-02-13:
+1. That is awesome. The underlying library is a bit old, but I can imagine
several ROS-I uses.  It might be worth utilizing. Note the MPL license. I'm
not too familiar with that license.

sketchc89, 2017-02-13:
MPL is [Mozilla Public License](https://www.mozilla.org/en-US/MPL/2.0/FAQ/). It's in between GPL and MIT/BSD. You need to distribute source of modified MPL code but not the projects that contain that code (not a lawyer).

 You might not need the libraries linked if that's a problem and could use other implementations or re-implement the algorithms from the pseudo-code on wikipedia since they're commonly used algorithms in graphics.

closed 2017-03-09 by AustinDeric

Jmeyer1292, 2017-03-09:
Note that edge path density is now controlled by the voxelization of input clouds. There's a 1.5mm spacing now, but this is still quite small. Our edge paths are 10x the density of the other paths. If in the future we encounter dwelling or we want to speed up motion planning, this may be relevant.

PR #189 (2017-03-09): VoxelGrid Filter
Filter the initial pointcloud through a pcl voxelgrid filter using the leaf size method (currently set at 1.5mm).

Fixes #151