ros-industrial-attic/godel #233: problems linking noether libraries?

JoePolden, 2017-07-03:
There seems to be some issue related to linking the vtk_viewer library to the godel_noether package. When I clone/build/link the noether meshing plugins into the godel workspace (following the instructions in the readme), i get the following errors when when I run the most generic demo:

	roslaunch godel_irb2400_support irb2400_blending.launch

results in:

	[ INFO] [1499073092.429216427]: Initializing godel_noether::NoetherMesher
	[ INFO] [1499073092.430949004]: godel_noether::NoetherMesher::generateMesh
	[ INFO] [1499073094.624346434]: Adding a marker for mesh with 3786 points
	[ INFO] [1499073094.625000896]: Initializing godel_noether::NoetherMesher
	[ INFO] [1499073094.626014572]: godel_noether::NoetherMesher::generateMesh
	[ INFO] [1499073095.351356182]: Adding a marker for mesh with 13248 points
	[ INFO] [1499073095.352588848]: Initializing godel_noether::NoetherMesher
	[ INFO] [1499073095.353504740]: godel_noether::NoetherMesher::generateMesh
	[ INFO] [1499073096.242733336]: Adding a marker for mesh with 12198 points
	[ INFO] [1499073096.506464312]: Created record with id: 1
	[ INFO] [1499073096.789307737]: Created record with id: 2
	[ INFO] [1499073097.159198151]: Created record with id: 3
	[ INFO] [1499073122.968884912]: Process Planning goal just went active
	[ INFO] [1499073123.026015460]: Starting Noether path planning...
	================================================================================REQUIRED process [surface_blending_service-11] has died!
	process has died [pid 23224, exit code -11, cmd /home/ros/vtk_noether_debugging_ws/devel/lib/godel_surface_detection/surface_blending_service __name:=surface_blending_service __log:=/home/ros/.ros/log/863ac282-5fcf-11e7-90cc-d481d76220b4/surface_blending_service-11.log].
	log file: /home/ros/.ros/log/863ac282-5fcf-11e7-90cc-d481d76220b4/surface_blending_service-11*.log
	Initiating shutdown!
	================================================================================

I tracked the errors to the vtk_viewer library, where accessing any vtkPolyData object will crash the node. However the fact that I am able to test and run these same libraries sucsessfully from inside my godel workspace by running 

	roslaunch noether noether.launch filename:=/path/to/unzipped/godel_point_cloud_data/surface3.pcd

leads me to beleive that it is an issue related to linking the vtk_viewer library to the godel_noether package.
Is anyone else able to recreate this error? Also, if there is a more appropriate place to direct these queries, please let me know. 

Cheers.

p.s. I am using VTK 8.1 and PCL 1.8 (which has been compiled against VTK 8.1)

akgoins, 2017-07-05:
Do you have pcl_ros and pcl_conversions in your work space and are they set to compile against PCL 1.8?  If you have already done this, you could try to compile PCL in release mode.  I have found that there are other run time issues with PCL and compiling in release mode usually fixes these.

JP-SG-ROS, 2017-07-05:
Thanks for the comment, but unfortunately these do not seem to resolve the issue. Additionally, I am not convinced that the issue relates with PCL, but VTK rather, since the errors only come when accessing vtkPolyData objects passed from NoetherPathPlanner::generatePath() function into the vtk_viewer library. 

Frustratingly, though, these errors do not occur when passing the same vtkPolyData object into the vtk_viewer library from the noether library (see my crude diagram below) in the same workspace, which leads me to think it is a linking error of some type. I will look into this a bit more. 

-------------------------------------------------- godel_ws ------------------------------------------------------------
 |   NoetherPathPlanner::generatePath() --->  vtkPolyData Obj ---> vtk_viewer library  ==  ERROR! |
 |                                                                                                                                                 |
 |   noether::main() --->  vtkPolyData Obj  ---> vtk_viewer library  == Works!                                   |
---------------------------------------------------------------------------------------------------------------------------

akgoins, 2017-07-05:
Okay, I thought that PCL was the problem, since pcl_ros is a wrapper around PCL 1.7 (Ubuntu 16.04/kinetic), which uses VTK 6.2.  If you include pcl_ros and the Noether library, your catkin workspace is now trying to include VTK 6.2 and 7.1.  Thus, the noether library would work with itself, because it uses VTK 7.1 and does not depend on pcl_ros.  The Godel packages, however, include pcl_ros, which can introduce this compile/run issue unless you build it from source and make it compile against PCL 1.8 (which should be built on VTK 7.1 already).  If you have already done this, then you can ignore what I just said.

I was looking at the godel_noether package and found something which might be a bug. The `generateNormals` line [here](https://github.com/Jmeyer1292/godel_noether/blob/master/src/noether_path_planner.cpp#L302) should be before the line directly above it where the mesh and normals get encoded together.  I am not sure how the Noether library would respond to having a mesh without normals.

AustinDeric, 2017-07-05:
Please note that we have docker images which can help in this case to standardize environments.  Y'all might want to look at debugging using this docker images:
https://hub.docker.com/r/rosindustrial/noether/

Jmeyer1292, 2017-07-05:
> I was looking at the godel_noether package and found something which might be a bug. The generateNormals line here should be before the line directly above it where the mesh and normals get encoded together. I am not sure how the Noether library would respond to having a mesh without normals.

@akgoins if that's a bug in `godel_noether` then it's a bug in [noether](https://github.com/ros-industrial/noether/blob/master/noether/src/noether.cpp#L187) as well. Admittedly, the whole "normals" situation is all kinds of messed up here, but the first call is generating normals from a point cloud of mesh vertices. The second call takes those point normals and computes `cell` (face?) normals.

JP-SG-ROS, 2017-07-06:
> Okay, I thought that PCL was the problem, since pcl_ros is a wrapper around PCL 1.7 (Ubuntu 16.04/kinetic), which uses VTK 6.2. If you include pcl_ros and the Noether library, your catkin workspace [...]

You could well be onto something here. I will explore this more. Thanks for the explanation.

> Please note that we have docker images which can help in this case to standardize environments. Y'all might want to look at debugging using this docker images:

Definitely will do. Thanks!

JP-SG-ROS, 2017-07-31:
> Okay, I thought that PCL was the problem, since pcl_ros is a wrapper around PCL 1.7 (Ubuntu 16.04/kinetic), which uses VTK 6.2. If you include pcl_ros and the Noether library, your catkin workspace [...]

I recently revisited this issue and did a bit of tinkering and it turns out this comment from akgoins was 100% spot on the money. So I guess we can close this issue. Thanks so much for your input.

For those who are experiencing similar issues with the godel_noether library. I resolved the issue by cloning the perception_pcl (kinetic-devel branch) and pcl_conversions packages into my workspace. Then, in the pcl_ros package housed inside the perception_pcl folder, I altered the CMakeLists.txt file by changing

`find_package(PCL 1.7 REQUIRED) `  to  -->  `find_package(PCL 1.8 REQUIRED)`

After this, the noether plugins worked. Thanks again for your help!