ros-industrial-attic/godel #234: run into problems compiling godel

mitch1044444, 2017-07-07:
Hey all,

I want to learn from your milestone 3 project. I followed your instruction to clone and compile, I can't use catkin build, it would give error "Could not find a package configuration file provided by "manipulation_msgs"
". 

I tied catkin_make_isolated , and everything, and again.
`<== Failed to process package 'godel_process_execution': 
  Command '['/home/drpslab/scann1/devel_isolated/swri_profiler/env.sh', 'cmake', '/home/drpslab/scann1/src/godel/godel_process_execution', '-DCATKIN_DEVEL_PREFIX=/home/drpslab/scann1/devel_isolated/godel_process_execution', '-DCMAKE_INSTALL_PREFIX=/home/drpslab/scann1/install_isolated', '-G', 'Unix Makefiles']' returned non-zero exit status 1
`
I wonder if this repository is working? I know you guys on working on the milestone 4, look forward to it.

geoffreychiou, 2017-07-08:
Do you have manipulation_msgs installed?

I believe milestone 3 is running on indigo so the command is:
```
sudo apt-get install ros-indigo-manipulation-msgs
```

Thompson104, 2017-07-08:
Thank you for your reply. I just checked the indigo version and it works well. 

Thank you so much for your efforts.