ros-industrial-attic/godel #236: questions on how to generate "godel" trejectory

Thompson104, 2017-07-12:
Hey all,

I'm not sure if this is the right place to ask this question. I wonder if you could provide some sample code or tutorial on how to generate similar trajectory like this

[image]

I checked your code but it is not that user friendly, I guess it's my problem. But I guess a good document would help me and others in the future too. 

Really appreciate any help. Thank you.

Jmeyer1292, 2017-07-12:
> user friendly

Yeah...

We used an external library to do this: https://github.com/aewallin/openvoronoi. I don't know too many of the details, but could point you to the right code if you need it.

Thompson104, 2017-07-13:
@Jmeyer1292 I'm actually a graduate student in mechanical engineering. I knew openvoronoi and I just don't know how to use this library :). 

Could you provide some suggestions on how to get a surface from PCL and then generate this kind of trajectory?  Or is there any library similar to openvoronoi that is not too difficult to understand?