ros-industrial-attic/godel #241: Compile Error

trent-weiss, 2018-05-24:
Hi Guys!

  I am having some trouble compiling the godel code... I get a compile error in generate_motion_plan.cpp saying:
``
```
/home/trent/catkin_dev_ws/src/godel/godel_process_planning/src/generate_motion_plan.cpp:79:47: error: no matching function for call to ‘descartes_planner::DAGSearch::run(std::vector<double>&)’
   double cost = search.run(process_start_costs);
                                               ^
In file included from /home/trent/catkin_dev_ws/src/godel/godel_process_planning/src/generate_motion_plan.cpp:7:0:
/home/trent/moveitbuild/install/include/descartes_planner/ladder_graph_dag_search.h:36:10: note: candidate: double descartes_planner::DAGSearch::run()
   double run();
          ^
/home/trent/moveitbuild/install/include/descartes_planner/ladder_graph_dag_search.h:36:10: note:   candidate expects 0 arguments, 1 provided
make[2]: *** [CMakeFiles/godel_process_planning_node.dir/src/generate_motion_plan.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/godel_process_planning_node.dir/all] Error 2
make: *** [all] Error 2

```

Also note I compiled Descartes from source on the kinetic-devel branch.

Has anyone else had this issue/know how to fix it?

Thanks,
Trent

Jmeyer1292, 2018-05-24:
Use the following branch of Descartes (sorry)
https://github.com/ros-industrial-consortium/descartes/tree/godel_refactor_omp

trent-weiss, 2018-05-25:
that seems to fix the compile issue, so I'll go ahead and close this.

But it seems like a bad idea to have two divergent branches of descartes serving two different purposes in the same repo...

closed 2018-05-25 by trent-weiss