ros-industrial-attic/godel #244: Status of Godel Project?

melvinma, 2018-09-13:
It appears that this project has been inactive for half a year now. Could someone summarize what is going on? Milestone 4 achieved and no further plan? Anything else a new comer needs to know? Thanks.

melvinma, 2018-09-15:
This is how I was able to successfully compile this code -

- as previously mentioned, there needs a small change to avoid comflict with libsocket++ installed in OS;
- Switch Discartes

```
original:
     git: {local-name: descartes, uri: 'https://github.com/Jmeyer1292/descartes.git', version: godel_refactor_omp}
new:
     git: {local-name: descartes, uri: 'https://github.com/ros-industrial-consortium/descartes.git', version: kinetic_devel}
```

- need to change at least two files because of Discartes change in the new source -

```
file: descartes/descartes_planner/include/descartes_planner/ladder_graph_dag_search.h
         original:  double run();
         new:       double run(const std::vector<double>& seed_weights = {});
```

```
file: descartes/descartes_planner/src/ladder_graph_dag_search.cpp
         replace funtion run() to run(const std::vector<double>& seed_weights) from        https://github.com/Jmeyer1292/descartes/blob/godel_refactor_omp/descartes_planner/src/ladder_graph_dag_search.cpp
```

robinsonmm, 2018-09-19:
Thanks for your inquiry. Godel is still of interest and we are working scope and interested parties for M5. We are also pointing other parties to this repository for additional leverage, so hopefully something will happen on that front. Unfortunately we cannot be very deterministic on timing, but rest assured, things are happening!

wilcobonestroo, 2020-03-31:
We are also interested in the status of this project. Based on the activity in the repo, it looks like there aren't people working on this any more. Based on the messages above from @melvinma , we were able to build the project. We might want to use it in a ROS training. However, we would like to use ROS Melodic for that. Are there plans to port the project to Melodic?

gavanderhoorn, 2020-04-06:
I'm not aware of any direct plans to port it to Melodic as of today.

However, several forks seem to include the necessary changes already.

If you're after seeing what changes would be necessary, comparing those forks with this repository would seem sufficient.

If you're wondering whether `godel` will see future development (ie; is not a 'dead project'), that I can't say at this point in time.