ros-industrial-attic/godel #246: Broken link in instructions

DanWilly, 2019-02-19:
Hey,

I just found this project and find it very interesting. I was trying to follow the instructions to run the simulation, sadly the link to the point cloud does not work anymore. Is that point cloud still available somewhere?

Details:
This XML file does not appear to have any style information associated with it. The document tree is shown below.
        
<Error><Code>AllAccessDisabled</Code><Message>All access to this object has been disabled</Message><RequestId>EEDD166780CE79C2</RequestId><HostId>0aEhEDs0NJ1B1gneVJWnNhl/6Hvi9JGL9XQkBfNNNLgWrtzn4q36oN2A6BxmEXT131g6gJQOSSo=</HostId></Error>

Thank you!

Regards

Daniel

jrgnicho, 2019-05-15:
It looks like the aws instance that holds those files has been permanently removed.

gavanderhoorn, 2019-05-15:
@jrgnicho: only disabled or has the bucket actually been removed / taken down?