ros-industrial-attic/godel #247: Compilation error for catkin_make?

shobhitcito, 2020-01-24:
Error:

CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_workspace.cmake:95 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:67 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/ra-om1/overlay_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/ra-om1/overlay_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

AustinDeric, 2020-01-28:
I'll try to reprodice this error, see PR #248.

with that said, it does appear there is an error here: 
```
godel_irb1200/abb_irb1200_5_90_ikfast_manipulator_plugin/src/abb_irb1200_5_90_manipulator_ikfast_moveit_plugin.cpp:366:102: error: no matching function for call to ‘const_pointer_casturdf::Link(urdf::LinkConstSharedPtr)’
```

shobhitcito, 2020-01-30:
@AustinDeric  thanks for looking into it

I am still not able to resolve the problem. I tried to go through the issues faced by other users and it hasn't helped me yet.

What should I do now? Thank you very much for helping out.

sumr4693, 2020-05-02:
Hi, was anyone able to resolve the above issue when doing 'catkin_make_isolated'? I am facing the same issue. 

Thanks in advance!

sshakthibalan, 2020-08-09:
I am facing the exact same issue as above and I am using ROS melodic. Could anyone please solve this issue.

kvnptl, 2021-06-02:
I am also facing the same error on ROS Melodic

kvnptl, 2021-06-03:
At least after some debugging, godel packages are now building without any errors (still few warnings are there) on my system (ROS Melodic).

I'll try to list out few changes/resolutions here:
1. Use `catkin build` tools instead of `catkin_make` or anything else. (delete build and devel, if you already built the package with `catkin_make`)
2. Replace `boost` with `std` in IKFastPlugin : https://github.com/ros-industrial/fanuc/issues/241
3. For error: `error: static assertion failed: YOU_PERFORMED_AN_INVALID_TRANSFORMATION_CONVERSION
     #define EIGEN_STATIC_ASSERT(X,MSG) static_assert(X,#MSG);`  => **Resolution**: https://github.com/ros-industrial-consortium/descartes/issues/234
4. Change logError, logInform,etc... with ROS_ERROR, ... [check this](https://answers.ros.org/question/322630/logerror-was-not-declared-in-this-scope/?answer=379471#post-id-379471)
5. For error: `error: conflicting declaration ‘typedef struct LZ4_stream_t LZ4_stream_t’
typedef struct { long long table[LZ4_STREAMSIZE_U64]; } LZ4_stream_t;`  => **Resolution**: https://github.com/ethz-asl/lidar_align/issues/16#issuecomment-758960226
6. For error: `error: ‘vector’ in namespace ‘std’ does not name a template type  typedef std::vector<Eigen::Affine3d, Eigen::aligned_allocator<Eigen::Affine3d> > PoseVector;` => **Resolution**: add `#include <vector>` in corresponding error file
7. For error: `error: ‘accumulate’ is not a member of ‘std’
     auto cost = std::accumulate(delta_buffer_.cbegin(), delta_buffer_.cend(), 0.0);` => **Resolution**: https://github.com/ros-industrial-consortium/descartes/pull/220/commits/fdfb598c422de04f0f1e08021ff9cd9ecc6b8083
8. For error: `undefined reference to `tf::poseMsgToEigen(geometry_msgs::Pose_<std::allocator<void> > const&, Eigen::Transform<double, 3, 1, 0>&)'` =>Resolution: add `eigen_conversions` inside `find_package(catkin REQUIRED COMPONENTS )` in `CMakeLists.txt. `

NOTE: still haven't tried demo launch files

Please comment if further explanation required or clarification.

gavanderhoorn, 2021-06-03:
Looking at forks of a repository is always very worthwhile.

The [tecnalia-advancedmanufacturing-robotics/godel](https://github.com/tecnalia-advancedmanufacturing-robotics/godel) and [sam-xl/godel](https://github.com/sam-xl/godel) forks seem to have resolved many (if not all) of the issues you found.